mikepurvis/ros-install-osx #116: macOS High Sierra installation

neuropacabra, 2018-03-01:
Hi,

I have tried to install ROS Kinetic Kame with this tutorial about 6 months ago. Is there any progress?

After hours of building I haven't managed to install ROS at all. I was only able to run roscore but anything else (RViz, rqt etc.) didn't work. It would be very pleasant for me to run ROS natively, now I run it in VM and it is not very great solution.

My question is, is there any working tutorial to install it? And does it working?

Back in time when I was trying to install ROS build failed with matplotlib and Qt, but I had the right versions and proper paths in terminal.

Thank you,

Michael

grafoteka, 2018-03-26:
Hi @neuropacabra, 
I'm in the same point as you. I try the installation last year, and now again and I have no good results, so I have to continue using ubuntu in a partition.

neuropacabra, 2018-03-27:
Hi @grafoteka  it is great to hear :-) i mean it. I read all over the place that everyone can run ROS under macOS High Sierra, but I have actually never seen anyone around me to successfully run it. One way (I am thinking about) is Docker. But even with Docker I had no luck. Unfortunately I have latest Macbook Pro 2017 which has basically no Linux support (no susspension, no audio, no inbuild peripherals etc.)

quinnabrvau, 2018-03-27:
This dockerfile will easily run the command line options but no GUI 
```# install ros
FROM ros:kinetic-ros-base-xenial
RUN apt-get update
RUN apt-get install -y ros-kinetic-perception=1.3.1-0*
RUN apt-get install -y nano
RUN rm -rf /var/lib/apt/lists/*

# add ros path to bash profile
RUN echo "source /opt/ros/kinetic/setup.bash" >> /root/.bashrc

# create catkin workspace
RUN /bin/bash -c 'source /opt/ros/kinetic/setup.bash && \
    mkdir -p /root/catkin_ws/src && cd /root/catkin_ws && catkin_make && \
    echo "source /root/catkin_ws/devel/setup.bash" >> /root/.bashrc'```

horagong, 2018-04-26:
I am using it natively on mac High Sierra with Rviz, Rqt, Moveit. It's possible but you will have to look around issues of each git repo.

neuropacabra, 2018-04-26:
@horagong Can you please guide me a little? This is what I tried so far. It did not work for me at all.[
[ROS install macOS.txt](https://github.com/mikepurvis/ros-install-osx/files/1951597/ROS.install.macOS.txt)
](url)

horagong, 2018-04-27:
@neuropacabra Did you compile it without errors? If so, try this.
* ros_comm is broken in recent version. (https://github.com/ros/ros_comm/issues/1357)
So I am using commit 46d5b. In this version comment out `close_half_closed_sockets()`
```
--- a/tools/rosmaster/src/rosmaster/util.py
+++ b/tools/rosmaster/src/rosmaster/util.py
@@ -65,7 +65,7 @@ def xmlrpcapi(uri):
         return None
     if not uri in _proxies:
         _proxies[uri] = ServerProxy(uri)
-    close_half_closed_sockets()
+    #XXX close_half_closed_sockets()
     return _proxies[uri]
```
* In urdf file, change shared library path into /opt/ros/lunar/lib. `.so` is ok but the path should be changed to abs path for loading.
```
  <gazebo>
    <plugin name="gazebo_ros_control" filename="/opt/ros/lunar/lib/libgazebo_ros_control.so">
      <robotNamespace>/</robotNamespace>
    </plugin>
  </gazebo>
```

berkaydeniz, 2018-04-30:
I would highly recommend installing Lunar, which works almost out-of-the-box.

neuropacabra, 2018-05-01:
@berkaydeniz OK, finally I got into complitation :-) I stucked on `rostime` package which has `boost` dependency. I installed `boost` with `brew install boost` and `brew install python-boost` and event downloaded and compiled from source. Even then I end up with `TypeError: coercing to Unicode: need string or buffer, NoneType found` when passing `boost` into terminal. I cannot solve this. Can you give a hint what to do?

berkaydeniz, 2018-05-01:
@neuropacabra, unfortunately I don't have any advice on this matter because I didn't have such a problem. One grave advice I can give you is, in order for this to work with little to no problem, you kinda need to wipe out everything you installed before. Last time I used this script, I started with a fresh OS X installation. 

The correct version of boost should be installed by the script itself. The problem you are having is potentially due to a previous installation of the library messing things up but this is just speculation.

neuropacabra, 2018-05-01:
@berkaydeniz Oh, I was thinking about this 👍 I will try in virtual and then I will write a report in case anyone would find it useful. Thanks a lot!

IcyZKunGz, 2018-05-02:
> @neuropacabra, unfortunately I don't have any advice on this matter because I didn't have such a problem. One grave advice I can give you is, in order for this to work with little to no problem, you [...]

> @berkaydeniz Oh, I was thinking about this 👍 I will try in virtual and then I will write a report in case anyone would find it useful. Thanks a lot!

I have just tried to install ros in Parallels Desktop with clean High Sierra install and it didn't work. Error still be rostime and it is the same error as I saw before.

Sorry for my poor English.

neuropacabra, 2018-05-02:
@IcyZKunGz I have same experience.

I was able to cpmpile like 36 out of 250 packages. I always endup with `rostime` errors I described eralier. I think it's `boost` related.

Without `rostime` all packages then fail. So I ended up my ROS on macOS desire. I get another machine and enjoy my life again :-)

NcJie, 2018-05-03:
@neuropacabra I ran into the same problem, I was able to compile it with boost 1.63. Ros-lunar does not seem to be compatible with boost 1.67 
1. Uninstall / unlink the latest boost version (1.67) 
    - `brew uninstall boost`
    - `brew unlink boost`
2. Install boost 1.63 
    - `brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/2898efab0f331be9663a640c2fae8b191b9538b2/Formula/boost.rb`   
    - `brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/a82238a3936a2691a17a516fc1fd6c67055160fa/Formula/boost-python.rb`
    - The formula link is obtained from `https://github.com/Homebrew/homebrew-core/commits/master/Formula/boost.rb`. I selected the updated 1.63 commit 
    - Note that, Xcode 9.3 is required 
3. Re-run `install.sh`

grafoteka, 2018-05-03:
@NcJie I have try your tips, but I have the next error:

Errors     << rostime:make /Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/logs/rostime/build.make.000.log                                                            
In file included from /Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/duration.cpp:35:
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/include/ros/impl/duration.h:182:31: error: no matching conversion for functional-style cast from 'double' to 'bt::microseconds' (aka 'subsecond_duration<boost::posix_time::time_duration, 1000000>')
    return bt::seconds(sec) + bt::microseconds(nsec/1000.0);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/duration.cpp:75:18: note: in instantiation of member function 'ros::DurationBase<ros::Duration>::toBoost' requested here
  template class DurationBase<Duration>;
                 ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'double' to 'const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000>' for 1st argument
  class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                             ^
/usr/local/include/boost/date_time/time_duration.hpp:286:59: note: candidate template ignored: disabled by 'enable_if' [with T = double]
                                typename boost::enable_if<boost::is_integral<T>, void>::type* = 0) :
                                                          ^
In file included from /Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/duration.cpp:35:
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/include/ros/impl/duration.h:182:31: error: no matching conversion for functional-style cast from 'double' to 'bt::microseconds' (aka 'subsecond_duration<boost::posix_time::time_duration, 1000000>')
    return bt::seconds(sec) + bt::microseconds(nsec/1000.0);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/duration.cpp:76:18: note: in instantiation of member function 'ros::DurationBase<ros::WallDuration>::toBoost' requested here
  template class DurationBase<WallDuration>;
                 ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'double' to 'const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000>' for 1st argument
  class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                             ^
/usr/local/include/boost/date_time/time_duration.hpp:286:59: note: candidate template ignored: disabled by 'enable_if' [with T = double]
                                typename boost::enable_if<boost::is_integral<T>, void>::type* = 0) :
                                                          ^
2 errors generated.
make[2]: *** [CMakeFiles/rostime.dir/src/duration.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/time.cpp:41:
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/include/ros/impl/time.h:170:35: error: no matching conversion for functional-style cast from 'double' to 'pt::microseconds' (aka 'subsecond_duration<boost::posix_time::time_duration, 1000000>')
    return pt::from_time_t(sec) + pt::microseconds(nsec/1000.0);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/time.cpp:590:18: note: in instantiation of member function 'ros::TimeBase<ros::Time, ros::Duration>::toBoost' requested here
  template class TimeBase<Time, Duration>;
                 ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'double' to 'const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000>' for 1st argument
  class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                             ^
/usr/local/include/boost/date_time/time_duration.hpp:286:59: note: candidate template ignored: disabled by 'enable_if' [with T = double]
                                typename boost::enable_if<boost::is_integral<T>, void>::type* = 0) :
                                                          ^
In file included from /Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/src/time.cpp:41:
/Users/jorge/Downloads/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/rostime/include/ros/impl/time.h:170:35: error: no matching conversion for functional-style cast from 'double' to 'pt::microseconds' (aka 'subsecond_duration<boost::posix_time::time_duration, 1000000>')
    return pt::from_time_t(sec) + pt::microseconds(nsec/1000.0);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
                             ^
/usr/local/include/boost/date_time/time_duration.hpp:286:59: note: candidate template ignored: disabled by 'enable_if' [with T = double]
                                typename boost::enable_if<boost::is_integral<T>, void>::type* = 0) :
                                                          ^
3 errors generated.
make[2]: *** [CMakeFiles/rostime.dir/src/time.cpp.o] Error 1
make[1]: *** [CMakeFiles/rostime.dir/all] Error 2
make: *** [all] Error `2`

And I can only compile 33 packages.

plguhur, 2018-05-09:
For the `tf2` error, you need to apply this patch: `https://github.com/ros/geometry2/pull/279/files` (I suppose you are building `lunar`)

asimonov, 2018-05-09:
did anyone build ROS (kinetic or lunar) on Mac OS using anaconda's environment rather than 'brew' python?

asimonov, 2018-05-09:
re: above comment. I have given up on trying to use 'conda' env with python 2.7. reinstalled brew and now trying to make the `install` script run from scratch.
again, seeing `boost 1.67` issue.
is there a plan to change main script to use boost 1.63?

mikepurvis, 2018-05-09:
Can Homebrew let you install versions other than the current one? I thought you could pretty much just switch between linking the different versions you happened to already have present on your machine.

In any case, my preference would be to see the issue with Boost 1.67 resolved.

asimonov, 2018-05-09:
right, I think it is enough to just have boost 1.63. boost-python that I have is 1.67 and i can go past that boost compile error

asimonov, 2018-05-09:
i can install older versions of boost and boost-python as suggested by @NcJie . but boost-python does not install. shows the error highlighed above

asimonov, 2018-05-09:
I managed to install (semi-manually) boost-python/boost 1.64. now I get the following, which relates to Apple Clang, as it seems:

```
Errors     << camera_calibration_parsers:make /Users/alexeysimonov/dev/ros-install-osx/lunar_desktop_full_ws/logs/camera_calibration_parsers/build.make.000.log                                                                                                                           
In file included from /Users/alexeysimonov/dev/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:38:
In file included from /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/yaml.h:10:
/usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/parser.h:41:3: warning: explicit conversion functions are a C++11 extension [-Wc++11-extensions]
  explicit operator bool() const;
  ^~~~~~~~
In file included from /Users/alexeysimonov/dev/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:38:
In file included from /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/yaml.h:14:
/usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:100:47: error: a space is required between consecutive right angle brackets (use '> >')
    const T&, typename disable_if<is_numeric<T>>::type* = 0) {
                                              ^~
                                              > >
/usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:112:50: error: a space is required between consecutive right angle brackets (use '> >')
    const T& key, typename enable_if<is_numeric<T>>::type* = 0) {
                                                 ^~
                                                 > >
/usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:123:11: error: exception specification of overriding function is more lax than base version
  virtual ~Exception() YAML_CPP_NOEXCEPT;

```

did anyone see/overcome this?

asimonov, 2018-05-10:
this is this issue: https://github.com/ros-perception/image_common/issues/78

asimonov, 2018-05-10:
fixed by using '      -DCMAKE_CXX_STANDARD=14 ' in catkin config

asimonov, 2018-05-10:
`pcl_ros` gives this strange error, which seems to be related to brew version of `pcl`

```
Errors     << pcl_ros:make /Users/alexey/dev/ros-install-osx/lunar_desktop_full_ws/logs/pcl_ros/build.make.001.log                                      
make[2]: *** No rule to make target `/usr/local/Cellar/python@2/2.7.14_3/Frameworks/Python.framework/Versions/2.7/Python', needed by `/Users/alexey/dev/ros-install-osx/lunar_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/pointcloud_to_pcd'.  Stop.
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
```

asimonov, 2018-05-10:
fixed by installing brew formula from python@2 2.7.14_3
and finally I have working lunar build on OS X!

plguhur, 2018-05-10:
Do you have any error about `std::vector` when you start `roscore`?

asimonov, 2018-05-10:
Nope

I had to use boost 1.64 from old brew formulas
This way all pieces of puzzle seem to fall together

_____________________________
Kind Regards,
Alexey Simonov

plguhur, 2018-05-10:
Could you send the details of your installation? I had installed `ros` but the puzzle fell apart when I launch `roscore`, even by trying some unmerged PRs.

asimonov, 2018-05-10:
I believe it is:

High Sierra 10.13.4
checked out this repo
reinstalled brew completely
installed/linked old brew python@2 formula for 2.7.14)3
installed/linked brew old boost and boost-python formulas for 1.64 (had to manually modify rb files for them to install)
then built the lunar workspace (just rebuilding it now from scratch to make sure it is using the right version of things)

neuropacabra, 2018-05-10:
@asimonov Is it working for you now?

asimonov, 2018-05-10:
nope. issue with ABI in rosout.

asimonov, 2018-05-11:
Yes, you can. The process is:

Find xxx.rb formula file on github
Look at its history for version you want
Brew install raw-url-of-formula

This works
But for older versions there are obviously references issues if they depend on other stuff

_____________________________
Kind Regards,
Alexey Simonov

neuropacabra, 2018-05-19:
Guys, anyone has some sort of complete guide?

asimonov, 2018-05-20:
Well,

I can summarise my notes. In the end a week or two ago i was able to build lunar on mac, but there was an issue in roscore itself that i was waiting to be resolved 

_____________________________
Kind Regards,
Alexey Simonov

horagong, 2018-05-21:
@asimonov As I said above, you can use `ros_comm 46d5b` commit for roscore workaround.
> ros_comm is broken in recent version. (ros/ros_comm#1357)
> So I am using commit 46d5b. In this version comment out close_half_closed_sockets()

neuropacabra, 2018-05-31:
Purely my curiosity...is Melodic more macOS firendly in terms of installation?

kgaranger, 2018-06-12:
I'm stuck with that `std::length_error`. Having read [ros/ros_comm#1357](https://github.com/ros/ros_comm/issues/1357) it seems to me that the issue was only solved for the lunatic distribution.
@horagong you mentioned that the older `ros_comm 46d5b ` commit could be used as a workaround, how to do that from the installation script?

berkaydeniz, 2018-06-14:
I decided to try a fresh install as I started having issues with ros message interfaces. Mainly, if a subscriber is registered before a publisher on the same topic, the subscriber callbacks are not being triggered when published messages are received.

Anyway, after going through a lot of hoops, and following various suggestions on this thread and googling for some more, I have a working roscore with the caveat of occasional error messages at the start. At least the message interfaces seem to work though so this might be yet another Mac OS X Ros gimmick we have to live with.

Here are my notes from the install:
- Completely removed homebrew, pip, and previous ros installation
- I had to modify the catkin config, and add `-DCMAKE_CXX_STANDARD=14` to avoid compiler errors triggered by `>>`
- on `rosdep.yaml`, as per the suggestion above, I added,
```
# Don't try to install google-mock
google-mock:
  osx:
    homebrew:
      packages: []
```
- the install script fails the first time and I have to install `python2`; run `brew install python2` manually, and restart the script
- installation of `gtest` fails at the end, due to conflicts with `yaml-cpp` on linking, had to force link it and restart the script
- the boost time errors mentioned above (`float` inputs instead of `uint32`) needs to be addressed at various places. each time I needed to do this is marked:
-  `roscpp_core` `uint32` fixes:
  - `roscpp_core/rostime/include/impl/duration.h`
  - `roscpp_core/rostime/include/impl/time.h`
- `xmlrpcpp` fixes from https://github.com/ros/ros_comm/pull/1392
- `tf2` fixes from https://github.com/ros/geometry2/pull/279
- `bondcpp` `uint32` fix: 
  - `bond_core/bondcpp/src/bond.cpp`
- for `cv_bridge` and `camera_calibration_parsers`, as suggested above, for `CMakeLists.txt`, I had to replace,
  - `find_package(Boost REQUIRED COMPONENTS filesystem python)` with,
  - `find_package(Boost REQUIRED COMPONENTS filesystem python27)`
- `actionlib` `uint32` fixes:
  - `actionlib/src/connection_monitor.cpp`
  - `actionlib/include/actionlib/destruction_guard.h`
  - `actionlib/include/actionlib/server/simple_action_server_imp.h`
  - `actionlib/include/actionlib/client/simple_action_client.h`
- `laser_assembler` `uint32` fixes:
  - `laser_assembler/test/test_assembler.cpp`
- `rviz` errors: `brew uninstall --ignore-dependencies --force qt@4`
  - I'm curious if anything else tried to use `qt4`
- `roscore` initialization error messages:
  - followed https://github.com/ros/ros_comm/pull/1393 as best as I can
  - the error messages still show up but at least the message interfaces seem to work fine.

berkaydeniz, 2018-06-14:
at some point, I will try to go through the same process with melodic instead of lunar

berkaydeniz, 2018-06-15:
Ok, as I have mentioned above, I have gone through the same process, this time with melodic. I simply modified the install script environment variable, cleaned up all pip and brew packages and started over. The result is a roscore that is error message free, and the message handling works properly. Here are my notes:
- once again, added `-DCMAKE_CXX_STANDARD=14` to test if it can avoid some of the silly issues
- once again, the script crashes complaining about `pip2`, simply `brew install python2` and start over with the script
- this time, I had to force link `yaml_cpp` over `gtest`: this must be a problem sooner or later, right?
- `bondcpp` `uint32_t` fixes
- `rosbag_storage` `openssl` missing header. Solution is based on https://stackoverflow.com/questions/45693149/openssl-aes-h-file-not-found-on-mac
  - First check if the path pointed out by ls $(brew --prefix openssl)/include exists.
  - If above is fine, you can try
```
cd /usr/local/include
ln -s ../opt/openssl/include/openssl .
```
- `actionlib` `uint32_t` fixes
- `cv_bridge` and `camera_calibration_parsers` require `python27` on their `CMakeLists.txt`
- `actionlib_tutorials` `uint32_t` (this is still `actionlib` fixes, this time with headers)
- `laser_assembler` `uint32_t` fixes

I hope this helps. I wonder if there is a more systematic way of addressing the float vs uint32 issue as it's on multiple packages.

berkaydeniz, 2018-06-15:
As an additional note, on `melodic`, `rviz` terminates gracefully!

mikepurvis, 2018-06-15:
@berkaydeniz Awesome, thank you for powering through with this! Please send a PR which makes the necessary modifications so that Melodic works. :)

asimonov, 2018-06-15:
That would be awesome to have it work on mac!!

_____________________________
Kind Regards,
Alexey Simonov

cmaclell, 2018-06-16:
Hey everyone, I'm trying to get Melodic to work on High Sierra and your notes have been incredibly helpful. 

What is the best way that you've all found to fix all the float vs uint32? Are you just manually editing the downloaded source files and then rerun the catkin build (without rerunning the ros-install-osx ./install script which rebuilds the workspace overwriting any changes I've made to the source files)?

berkaydeniz, 2018-06-16:
@cmaclell after the install reached the catkin build phase, after every error, I corrected the problems, and ran `catkin build` again.

@mikepurvis I will properly create forks with fixes to be included in the modified install script, test it and create a PR as soon as I can.

cmaclell, 2018-06-16:
@berkaydeniz Thanks. Also, you mention adding `python27` to the `CMakeLists.txt` for `cv_bridge` and `camera_calibration_parsers`. Where are the `CMakeLists.txt` files located? There does not seem to be folders for cv_bridge and camera_calibration_parsers in melodic_desktop_full_ws/src, which is where I expected them to be.

raskolnikov-reborn, 2018-07-04:
@berkaydeniz Hi. I would like to collaborate on getting melodic to work on high Sierra with you. Can you send me a message and lets divide up the workload and prepare a wiki/doc to get stuff moving?

meyerj, 2018-08-02:
For the record:
- https://github.com/ros-perception/image_common/pull/85 and
- https://github.com/ros-perception/vision_opencv/pull/239

fix the boost_python cmake problem with Boost 1.67 as suggested in https://github.com/mikepurvis/ros-install-osx/issues/116#issuecomment-391388778. I did not find this thread before.

neuropacabra, 2018-08-04:
I volunteer in any case of testing and especially writing documentation - finalize a sh script.

horagong, 2018-08-04:
@kgaranger I didn't see your msg. You can search and find ros_comm's github repo from ros site. Git clone the source in your src dir in which you downdoad and compile ros distribution removing old ros_comm dir. Git check out 46d5b in cloned ros_comm dir and compile with the same compile command.
You can replace other packages to the recent packages like this if there are some issues

horagong, 2018-08-04:
I am wondering you guys are running well with the recent ros_comm. It is said that the connetion bug is fixed but when I tried even the simple turtle bot it didn't work. So I was running it with ros_comm 46d5b. It works well.
When I tried the rdcent melodic there was no difference. It didn't work. If I change the ros_comm to 46d5b in melodic I think it will work. https://github.com/ros/ros_comm/issues/1469

But I didn't try it because I couldn't compile moveit in melodic. https://github.com/ros-planning/moveit/issues/1003
It was compiled in lunar. 
https://github.com/ros-planning/moveit/issues/818
So I am using lunar with ros_comm to 46d5b.
But others seems to have no issue on that ros_comm. I am wondering.

asimonov, 2018-08-08:
hi! 
i have tried the PR, but ran commands from install script manually. 
had few hiccups, but i think it has worked. 

roscore can start ok.

cannot start rviz. should it be available just by running rviz?

PS
do you recommend removing homebrew before installation? I have started from existing homebrew install

berkaydeniz, 2018-08-08:
hey, your existing homebrew explains why you needed to run several commands manually, it will manually make you upgrade packages you have. 

I would recommend purging homebrew and trying again.

I think there are still two instances the script fails with homebrew telling you what to do. You simply run the given command and then run the install script again. One of these is that weird force linking of pyyaml and the other happens only when you start after purging homebrew. Especially once you reach catkin you shouldn't need to interfere.

roscore, message interfaces, rviz, gazebo, rqt_plot, rqt_tree are all working for me. as long as you source your .bash_profile, just rviz should be enough.

asimonov, 2018-08-08:
the pull request works for me!

I had to purge Homebrew to make it happen.

High Sierra 10.13.6

```
$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.1.0 (clang-902.0.39.2)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

grafoteka, 2018-10-05:
Hi all, 
finally I could install and compile ROS Lunar on OSX High Sierra. 

Unfortunately, I have the error:

`[ERROR] [1538702529.917543000]: [registerService] Failed to contact master at [Jorges-MacBook-Pro.local:11311].  Retrying...`

I have read this issue [#1357](https://github.com/ros/ros_comm/issues/1357) but my files are like that. 

Also, if I want to launch Gazebo, I have the next error:

```
dyld: Symbol not found: __ZNK5boost16re_detail_10670031cpp_regex_traits_implementationIcE17transform_primaryEPKcS4_
  Referenced from: /usr/local/opt/sdformat5/lib/libsdformat.5.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib
 in /usr/local/opt/sdformat5/lib/libsdformat.5.dylib
/opt/ros/lunar/lib/gazebo_ros/gzserver: line 41:  1980 Abort trap: 6           GAZEBO_MASTER_URI="$desired_master_uri" GAZEBO_MODEL_DATABASE_URI="$desired_model_database_uri" gzserver $final
[gazebo-1] process has died [pid 1909, exit code 134, cmd /opt/ros/lunar/lib/gazebo_ros/gzserver -e ode /Volumes/almacen/WS/catkin_ws/src/arm_gazebo/worlds/empty.world __name:=gazebo __log:=/Users/jorge/.ros/log/13dd8666-c83d-11e8-8dc2-c82a1416cda2/gazebo-1.log].
log file: /Users/jorge/.ros/log/13dd8666-c83d-11e8-8dc2-c82a1416cda2/gazebo-1*.log
```

Has anyone else had the same problem?

Thank you, 
Jorge

michtesar, 2018-10-05:
@grafoteka Hi! Awesome! Can you provide a way how did you installed?

grafoteka, 2018-10-06:
Hi @michtesar sure! I will try to explain you in the next lines.

I have an MacBook Pro early 2011, with 16GB RAM and SSD, and a fresh installation of OSX High Sierra.

I download this repo and start the installation with the script ./install.

Here are the next errors that I had:

- Error #1: 
You have not agreed to the Xcode license. Please resolve this by running

Solution: sudo xcodebuild -license accept

- Error #2: 
Error: Cask 'brew-cask' is unavailable: '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask/Casks/brew-cask.rb' does not exist.
Updating Homebrew...
==> Satisfying dependencies
==> Downloading https://dl.bintray.com/xquartz/downloads/XQuartz-2.7.11.dmg
######################################################################## 100.0%
==> Verifying SHA-256 checksum for Cask 'xquartz'.
==> Installing Cask xquartz
==> Creating Caskroom at /usr/local/Caskroom
==> We'll set permissions properly so we won't need sudo in the future.
Password:
==> Running installer for xquartz; your password may be necessary.
==> Package installers may write to any location; options such as --appdir are i
installer: Package name is XQuartz 2.7.11
installer: Installing at base path /
installer: The install was successful.
🍺  xquartz was successfully installed!
Log out and in to finalize XQuartz setup.

Solution: Log out and init session

- Error #3:
No such file or directory @ dir_chdir - /usr/local/Cellar
Warning: Bottle installation failed: building from source.
==> Downloading https://ftp.gnu.org/gnu/gdbm/gdbm-1.18.tar.gz
Already downloaded: /Users/jorge/Library/Caches/Homebrew/downloads/a9e9f67cd910f031ddd1073a68e57975ce6369e57728ee4ab7b57aaaa2c39807--gdbm-1.18.tar.gz
Error: An exception occured within a child process:
Errno::EPERM: Operation not permitted @ dir_s_mkdir - /usr/local/Cellar

Solution: 
[https://stackoverflow.com/questions/37232601/errnoeperm-operation-not-permitted-dir-s-rmdir/37233446#37233446](https://stackoverflow.com/questions/37232601/errnoeperm-operation-not-permitted-dir-s-rmdir/37233446#37233446)
`sudo chmod 1777 /usr/local`

- Error #4:
`./install: line 94: pip2: command not found`

Solution:
`brew install python2`

- Error #5:
```
Collecting pycurl
  Using cached https://files.pythonhosted.org/packages/e8/e4/0dbb8735407189f00b33d84122b9be52c790c7c3b25286826f4e1bdb7bde/pycurl-7.43.0.2.tar.gz
    Complete output from command python setup.py egg_info:
    Using curl-config (libcurl 7.54.0)
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/sr/bvv3t_fx6rzds8ps0y2p5l9w0000gn/T/pip-install-UAasPc/pycurl/setup.py", line 913, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/private/var/folders/sr/bvv3t_fx6rzds8ps0y2p5l9w0000gn/T/pip-install-UAasPc/pycurl/setup.py", line 582, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/private/var/folders/sr/bvv3t_fx6rzds8ps0y2p5l9w0000gn/T/pip-install-UAasPc/pycurl/setup.py", line 99, in __init__
        self.configure()
      File "/private/var/folders/sr/bvv3t_fx6rzds8ps0y2p5l9w0000gn/T/pip-install-UAasPc/pycurl/setup.py", line 316, in configure_unix
        specify the SSL backend manually.''')
    __main__.ConfigurationError: Curl is configured to use SSL, but we have not been able to determine which SSL backend it is using. Please see PycURL documentation for how to specify the SSL backend manually.
    
[... 72 lines cut ...]
	1	Uninstall / unlink the latest boost version (1.67)
```
brew uninstall boost
brew unlink boost
```
	2	Install boost 1.63 and boost-python 1.63
	◦	brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/2898efab0f331be9663a640c2fae8b191b9538b2/Formula/boost.rb —with-python
	◦	brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/a82238a3936a2691a17a516fc1fd6c67055160fa/Formula/boost-python.rb
	◦	The formula link is obtained from https://github.com/Homebrew/homebrew-core/commits/master/Formula/boost.rb. I selected the updated 1.63 commit
	◦	Note that, Xcode 9.3 is required
	3	Re-run install.sh

- - - - - - 
Once, you have compile the compile everything, if you have the error that:

`roscore: command not found`

After that, I was able to launch ros core and some packages, but with the problems that there were no connections to localhost. (I don't remember the correct error, but was similar)

Also, installing boost 1.63 I think that is the fail by which Gazebo 8 can't be started.

Hope this can help you.
Jorge

Then, you have to do a new catkin build.

grafoteka, 2018-10-19:
Hi all! I continue trying to install ROS in my mac.

Now, I'm having the _laser_assembler_ error.

I have read the solution from @berkaydeniz, but I don't understand exactly where to modify _unit32_t_

I change in line 138:

`  uint32_t result = RUN_ALL_TESTS();`
But still having the same problem. 

Can someone guide me with this error? 

Thank you, Jorge

berkaydeniz, 2018-10-19:
you can have a look at the change I was pointing at in the pull request:
https://github.com/berkaydeniz/ros-install-osx/blob/master/install
scroll down to line 170

grafoteka, 2018-10-19:
Thank you for your answer @berkaydeniz 

I found the difference in the line 120.

I try removing only the ´f´and not the decimal part.

```
    cloud_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000.0));

    cloud_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000));
```

[5 more replies]

closed 2019-08-01 by michtesar