mikepurvis/ros-install-osx #117: /usr/local/opt/python/bin/python2.7: bad interpreter

NikolausDemmel, 2018-03-30:
It seems homebrew changed how they deal with python (again...) and now `/usr/local/opt/python` points to python3 instead of python2...

```
$ rosmsg show sensor_msgs/Image
-bash: /opt/ros/lunar/bin/rosmsg: /usr/local/opt/python/bin/python2.7: bad interpreter: No such file or directory
$ ipython
-bash: /usr/local/bin/ipython: /usr/local/opt/python/bin/python2.7: bad interpreter: No such file or directory
```

It seems this broke all pip installed and and python ros packages.

Has anyone dealt with this? Is there a way without reinstalling all of ros and also all pip packages? If reinstalling is the only way, does anyone happen to know to how to list all pip packages that are user-installed, and those that are installed system-wide, and at the same time not list any ROS python packages and python packages from homebrew?

trainman419, 2018-04-04:
I'm on Indigo and I'm trying to reinstall a newer version of ROS and I'm running into the same issue.

trainman419, 2018-04-04:
I was able to resolve the issues I was having (and get /usr/local/opt/python/bin/python2.7) by installing homebrew's python2 package:

    brew install python2

Perhaps this should be added to the install script?

NikolausDemmel, 2018-04-04:
I think it is `python@2` now (`python2` might be an alias).

I have both `python@2` and `python` (which is python 3.6) installed (just reinstalled both) and my `/usr/local/opt/python` links to the python 3.6 , i.e. `/usr/local/opt/python/bin/python2.7` doesn't exist. I'm not sure why it would still be there on your setup. Could you do `brew reinstall python@2 python` and see if they are still there?

Note however that `/usr/local/bin/python` still refers to `python2.7` (I think due to some PEP requiring it).

I need python 3.6, since some other brew Formulae I have installed depend on it.

I resolved the issue manually by:
 * uninstalling all local pip packages from `~/Library/Python/2.7/lib/python/site-packages`
 * uninstalling all global pip packages not related to homebrew from `/usr/local/lib/python2.7/site-packages/`
 * wiping `/opt/ros`
 * reinstalling ROS, installing needed pip packages in the process

trainman419, 2018-04-05:
`brew list` shows that I have `python` and `python@2` installed, which seems to indicate that `python@2` is the official package name.

After reinstalling `python` and `python@2`,  `/usr/local/opt/python/bin/python2.7`, but pip2 and other tools that use python2.7 still work, so it seems like something has changed.

mikepurvis, 2018-04-05:
See some discussion in #85. We already specify the exact python executable, libs, and include dir at build time, so anything that properly uses `catkin_install_python` should get the shebang set correctly:

https://github.com/mikepurvis/ros-install-osx/blob/4e342f604960e291320e6acba62ff349747707d5/install#L182-L184

So the main thing then is adding a switch to pick which Python you want, and probably changing the default to be 3, to minimize friction with Homebrew.

NikolausDemmel, 2018-04-05:
> So the main thing then is adding a switch to pick which Python you want, and probably changing the default to be 3, to minimize friction with Homebrew.

Not sure if a switch to 3 by default minimizes friction. `python` and `pip` executables in `/usr/local/bin` from homebrew are now (again) python 2 (https://docs.brew.sh/Homebrew-and-Python). But optional Python 3 support would be great, in particular to indentify and fix issues in other packages, and to be ready for the next change in homebrew's python handling.

NikolausDemmel, 2018-04-05:
I think we can close this issue? It shouldn't affect new setups and for old setups manual reinstallation as detailed above is probably the sanest solution.

Python 3 support is already tracked in #85.

mikepurvis, 2018-04-05:
Sounds good. Thanks for the discussion.

closed 2018-04-05 by mikepurvis