mikepurvis/ros-install-osx #118: Issue to build catkin

plguhur, 2018-05-08:
I am failing to do the `catkin build`: I have the following message error:

> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Profile:                     default
> Extending:             [env] /usr/local/opt/qt
> Workspace:                   /Users/pierre-louis/kinetic_desktop_full_ws
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Source Space:       [exists] /Users/pierre-louis/kinetic_desktop_full_ws/src
> Log Space:          [exists] /Users/pierre-louis/kinetic_desktop_full_ws/logs
> Build Space:        [exists] /Users/pierre-louis/kinetic_desktop_full_ws/build
> Devel Space:        [exists] /Users/pierre-louis/kinetic_desktop_full_ws/devel
> Install Space:      [exists] /opt/ros/kinetic
> DESTDIR:            [unused] None
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Devel Space Layout:          linked
> Install Space Layout:        merged
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Additional CMake Args:       -DCATKIN_ENABLE_TESTING=1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DPYTHON_EXECUTABLE=/Users/pierre-louis/.pyenv/shims/python2 -DPYTHON_LIBRARY=/Users/pierre-louis/.pyenv/versions/ros/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/Users/pierre-louis/.pyenv/versions/ros/include/python2.7
> Additional Make Args:        None
> Additional catkin Make Args: None
> Internal Make Job Server:    True
> Cache Job Environments:      False
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Whitelisted Packages:        None
> Blacklisted Packages:        None
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Workspace configuration appears valid.
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> [build] Found '228' packages in 0.0 seconds.
> Warning: generated devel space setup files have been deleted.
> Starting  >>> catkin
> ___________________________________________________________________________________________________________________________________________________________________________________________________
> Errors     << catkin:cmake /Users/pierre-louis/kinetic_desktop_full_ws/logs/catkin/build.cmake.003.log
> /Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin/_setup_util.py: line 36: This file generates shell code for the setup.SHELL scripts to set environment variables: command not found
> from: can't read /var/mail/__future__
> import: delegate library support not built-in '' (X11) @@ error/import.c/ImportImageCommand/1295.
> import: delegate library support not built-in '' (X11) @@ error/import.c/ImportImageCommand/1295.
> import: delegate library support not built-in '' (X11) @@ error/import.c/ImportImageCommand/1295.
> import: delegate library support not built-in '' (X11) @@ error/import.c/ImportImageCommand/1295.
> import: delegate library support not built-in '' (X11) @@ error/import.c/ImportImageCommand/1295.
> import: delegate library support not built-in '' (X11) @@ error/import.c/ImportImageCommand/1295.
> /Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin/_setup_util.py: line 46: CATKIN_MARKER_FILE: command not found
> /Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin/_setup_util.py: line 48: syntax error near unexpected token `('
> /Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin/_setup_util.py: line 48: `system = platform.system()'
> Traceback (most recent call last):
>   File "/Users/pierre-louis/kinetic_desktop_full_ws/build/catkin/catkin_generated/generate_cached_setup.py", line 22, in <module>
>     code = generate_environment_script('/Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin/env.sh')
>   File "/Users/pierre-louis/kinetic_desktop_full_ws/src/catkin/cmake/../python/catkin/environment_cache.py", line 62, in generate_environment_script
>     env_after = ast.literal_eval(output.decode('utf8'))
>   File "/Users/pierre-louis/.pyenv/versions/2.7.15/lib/python2.7/ast.py", line 49, in literal_eval
>     node_or_string = parse(node_or_string, mode='eval')
>   File "/Users/pierre-louis/.pyenv/versions/2.7.15/lib/python2.7/ast.py", line 37, in parse
>     return compile(source, filename, mode, PyCF_ONLY_AST)
>   File "<unknown>", line 1
>     Could not write the output of '/Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin/_setup_util.py' to temporary file '/var/folders/ty/1lsvbm5d23jfrmrt1cqkzy080000gn/T//setup.sh.kbADeOlymD': may be the disk if full?
>                   ^
> SyntaxError: invalid syntax
> CMake Error at /Users/pierre-louis/kinetic_desktop_full_ws/src/catkin/cmake/safe_execute_process.cmake:11 (message):
>   execute_process(/Users/pierre-louis/.pyenv/shims/python2
>   "/Users/pierre-louis/kinetic_desktop_full_ws/build/catkin/catkin_generated/generate_cached_setup.py")
>   returned error code 1
> Call Stack (most recent call first):
>   cmake/all.cmake:186 (safe_execute_process)
>   CMakeLists.txt:8 (include)
> 
> 
> cd /Users/pierre-louis/kinetic_desktop_full_ws/build/catkin; catkin build --get-env catkin | catkin env -si  /usr/local/bin/cmake /Users/pierre-louis/kinetic_desktop_full_ws/src/catkin --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/Users/pierre-louis/kinetic_desktop_full_ws/devel/.private/catkin -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCATKIN_ENABLE_TESTING=1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DPYTHON_EXECUTABLE=/Users/pierre-louis/.pyenv/shims/python2 -DPYTHON_LIBRARY=/Users/pierre-louis/.pyenv/versions/ros/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/Users/pierre-louis/.pyenv/versions/ros/include/python2.7; cd -
> ...................................................................................................................................................................................................
> Failed     << catkin:cmake                        [ Exited with code 1 ]
> ....

I am using pyenv because I had no way to install rosdep otherwise.

plguhur, 2018-05-08:
Looks like I did something wrong. I started again and just got stuck at https://github.com/mikepurvis/ros-install-osx/issues/110

closed 2018-05-08 by plguhur

maximosipov, 2018-11-17:
If using pyenv the generated python2 shebang isn't working for some reason and script gets interpreted with bash:

head /Users/maxim/Downloads/ros-install-osx/lunar_desktop_full_ws/devel/.private/catkin/_setup_util.py
#!/Users/maxim/.pyenv/shims/python2
# -*- coding: utf-8 -*-

# Software License Agreement (BSD License)
#
# Copyright (c) 2012, Willow Garage, Inc.
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions

/Users/maxim/Downloads/ros-install-osx/lunar_desktop_full_ws/devel/.private/catkin/_setup_util.py
/Users/maxim/Downloads/ros-install-osx/lunar_desktop_full_ws/devel/.private/catkin/_setup_util.py: line 36: This file generates shell code for the setup.SHELL scripts to set environment variables: command not found
from: can't read /var/mail/__future__