mikepurvis/ros-install-osx #124: Catkin command not found

channeng, 2018-05-20:
Hi, I am facing an issue with `catkin config` command which runs after the successful `rosdep install`. 
``
#All required rosdeps installed successfully
~/ros-install-osx/lunar_desktop_full_ws
This sudo prompt is to create and chown /opt/ros/lunar.
Password:
./install: line 176: catkin: command not found
```

I did some tweaks to `rosdep install` to add flags for `--skip-keys gtest --skip-keys google-mock` from this PR: https://github.com/mikepurvis/ros-install-osx/issues/110, which I'm not sure could've affected the installation.

Wondering if there's something that I could have missed out here? Thanks.

akshayprsd, 2018-07-04:
Seems like I am facing something similar. Anyone find a solution?

kapaza, 2018-07-06:
Reinstall catkin_tools

pip2 install -U catkin_tools

Works for me

lanma, 2018-07-19:
Seems like I am facing something similar.

lanma, 2018-07-19:
I pass this issue by Installing catkin_tools from source

$ git clone https://github.com/catkin/catkin_tools.git
$ cd catkin_tools
$ pip install -r requirements.txt --upgrade
$ python setup.py install --record install_manifest.txt

After the steps, you have catkin command right now. 
then ./install again
Reference:
http://catkin-tools.readthedocs.io/en/latest/installing.html#installing-on-other-platforms-with-pip

PS: stuck in another issue..