mikepurvis/ros-install-osx #128: Mojave Melodic Installation

smnogar, 2018-10-10:
I am currently working through getting Melodic up and running on Mojave. The current sticking point is qt_gui_cpp. I have gotten this to compile fine for High Sierra, so I think this is Mojave specific. I figured I'd create this thread for anyone else working through Mojave issues.

```
Errors     << qt_gui_cpp:make /Users/steve/Documents/ros-install-osx/melodic_desktop_full_ws/logs/qt_gui_cpp/build.make.001.log           
Traceback (most recent call last):
  File "/opt/ros/melodic/share/python_qt_binding/cmake/sip_configure.py", line 51, in <module>
    config = Configuration()
  File "/opt/ros/melodic/share/python_qt_binding/cmake/sip_configure.py", line 18, in __init__
    [qmake_exe, '-query'], env=env, universal_newlines=True)
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 216, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
```

bgromov, 2018-10-24:
I just successfully installed Melodic on Mojave and had no issues compiling Qt part at all. Though I migrated the system from High Sierra and maybe some parts remained from there. I did reinstall SIP however. I installed version 4.19.8_5 from source with the default parameters and everything seems fine.

P.S. I had all kind of other issues, that I described here (quite dirty at the moment): https://gist.github.com/bgromov/23a74bbe846d965964b150080cb2d574

bgromov, 2018-10-26:
Not sure if it is related to Qt, but I am having issues with overscaled GUI controls in RViz and Gazebo, though 3D views are fine. I have reported the issue here: https://github.com/ros-visualization/rviz/issues/1306

Did anyone encounter this problem?

Tutorgaming, 2018-12-05:
> Not sure if it is related to Qt, but I am having issues with overscaled GUI controls in RViz and Gazebo, though 3D views are fine. I have reported the issue here: [ros-visualization/rviz#1306](https://github.com/ros-visualization/rviz/issues/1306)
> 
> Did anyone encounter this problem?

Let's have a look at this thread on Gazebo's Repo 
There're some workaround of this scaled GUI 

https://bitbucket.org/osrf/gazebo/issues/2531/gazebo-gui-rendering-is-scaled

PS. Do you have any gist or script that you used in order to install ROS on Mojave ?

smnogar, 2018-12-15:
I was able to resolve my problems by installing the package located at `/Library/Developer/CommandLineTools/Packages/package macOS_SDK_headers_for_macOS_10.14.pkg`. So for any of you out there getting weird errors on packages that worked in high Sierra but are failing in Mojave follow that step.

However, Gazebo is currently broken, so alas I continue to wait to move to Mojave.

smnogar, 2018-12-31:
Looks like Gazebo works again as of 9.6.0. I am still having problems getting pcl_ros to compile, but otherwise it seems like Mojave+ROS is ready to go.

closed 2018-12-31 by smnogar

Argzero, 2019-07-24:
@smnogar - are there any comprehensive instructions anywhere on how to set that up? I've had loads of issues just getting any version of ROS installed on my Mojave.