mikepurvis/ros-install-osx #133: ROS installation on Mojave seems to be stuck at tapping from ros/deps, Invalid Formula..

Kzernobog, 2019-03-06:
```
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb
Calling MacOS.prefer_64_bit? is disabled! There is no replacement.
Please report this to the ros/deps tap:
  /usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:15

Error: Cannot tap ros/deps: invalid syntax in tap!
```
The install script prints this to the screen after i run ```install```. ```XQuartz``` is installed and then when it tries to ```brew tap ros/deps```, it throes this error.

How do I fix this?

Evi5, 2019-03-08:
me too...
I am try with docker now
http://wiki.ros.org/docker/Tutorials/Docker

GiulioPettenuzzo, 2019-03-13:
the same error might be resolved here..
https://stackoverflow.com/questions/55064406/installation-of-ros-kinetic-in-mac-mojave/55140779#55140779

Zhenye-Na, 2019-03-13:
Also refer to this

https://github.com/ros/homebrew-deps/issues/36