mikepurvis/ros-install-osx #140: PyQt5 create problem with python2

Tarun22Sahu, 2019-09-08:
`brew install pyqt5 --with-python --without-python3` no longer works.
In updated brew these options has been deprecated. Now building the package qt_gui_cpp and similar could not find the  pyqt5. I also tried to compile all steps again with python3. but some packages and even catkin_make need python2. I am stuck in this deadlock. some package need library for python which are only available for python3 like pyqt5 and in other way, not all things compiled with python3, they require python2.

Please @mikepurvis look into this and suggest some solution.

dawnos, 2019-09-11:
Yes. I also meet this problem.

lrapetti, 2019-09-19:
It seems they have remove `Python2` from the pyqt5 Formula https://github.com/Homebrew/homebrew-core/commit/252af0b1093e0489574ff736bb2f57142e3d2054. Same happened with `sip` with https://github.com/Homebrew/homebrew-core/commit/bf34b31fcbc2f8944fa9ff347afb245333e8c592 that is also causing problems to me.

lrapetti, 2019-10-15:
I think you can check in https://github.com/mikepurvis/ros-install-osx/pull/141/files#diff-04c6e90faac2675aa89e2176d2eec7d8R12-R22 the pointer to the working commit of those dependencies.