mikepurvis/ros-install-osx #144: homebrew: command [brew install gazebo8] failed

fuying-wang, 2020-08-10:
Hi,

Thanks for your work.

When I ran the `./install` and got the error like this:

```
executing command [brew install gazebo8]
Updating Homebrew...
==> Installing gazebo8 from osrf/simulation
Warning: osrf/simulation/gazebo8 has been deprecated!
==> Downloading https://github.com/osrf/gazebo/commit/69fcced27c1df15719f626eb3dc4721540a1de63.diff?full_index=1
Already downloaded: /Users/wangfuying/Library/Caches/Homebrew/downloads/37bb2a163e94cc6061817826b794fbbac2f6e760ce4c48e27e6765ccf2a26a42--69fcced27c1df15719f626eb3dc4721540a1de63.diff
==> Downloading https://github.com/osrf/gazebo/commit/9c3fa7f66e5d8333b7b6bf5349e53eff2cb11bc0.diff?full_index=1
Already downloaded: /Users/wangfuying/Library/Caches/Homebrew/downloads/c97a443c10088500872cdaeccc94620be8cabd240736903c7bba8e7db4830a71--9c3fa7f66e5d8333b7b6bf5349e53eff2cb11bc0.diff
==> Downloading https://github.com/osrf/gazebo/commit/3555b9021931f404b39195f51af1b07c4ee0df9c.diff?full_index=1
Already downloaded: /Users/wangfuying/Library/Caches/Homebrew/downloads/ba4425018394fb6fe618c1bd393ddb7d2dba0225765c791216a3ddd0feeb75d7--3555b9021931f404b39195f51af1b07c4ee0df9c.diff
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_f
Already downloaded: /Users/wangfuying/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-8.6.0.tar.bz2
Already downloaded: /Users/wangfuying/Library/Caches/Homebrew/downloads/5950c34577ccf3e9f4b67ca950791ef44d89c700c5326dbc48e51afe6bcdb2de--gazebo-8.6.0.tar.bz2
==> Patching
==> Applying 69fcced27c1df15719f626eb3dc4721540a1de63.diff
patching file cmake/SearchForStuff.cmake
Hunk #1 succeeded at 568 (offset 17 lines).
==> Applying 9c3fa7f66e5d8333b7b6bf5349e53eff2cb11bc0.diff
patching file cmake/gazebo-config.cmake.in
Hunk #1 FAILED at 103.
1 out of 1 hunk FAILED -- saving rejects to file cmake/gazebo-config.cmake.in.rej
Error: Failure while executing; `patch -g 0 -f -p1 -i /private/tmp/gazebo8--patch-20200810-39696-pw82of/9c3fa7f66e5d8333b7b6bf5349e53eff2cb11bc0.diff` exited with 1.
ERROR: the following rosdeps failed to install
  homebrew: command [brew install gazebo8] failed
```

My system is `Mac OS Catalina version 10.15.6`.

Thanks for your help!

ragini-dwivedi, 2021-04-17:
I have the same error on MacOS Big Sur, version - 11.2.3

modanesh, 2021-06-14:
Same issue on macOS big sur 11.4.

anadminuser, 2022-01-03:
I also have this issue on Big Sur 11.6.1